ixmilia/dxf-rs #3: Unable to open several DXF files

Michael-F-Bryan, 2017-10-09:
I've got a couple DXF files which open just fine in a DXF viewer or SolidWorks, but when you try to load them with the `dxf` crate it fails with `the specified enum value does not fall into the expected range`.

Would you happen to know where the error comes from, or somewhere I should start looking if I want to write a fix?

[drawings.zip](https://github.com/IxMilia/dxf-rs/files/1367402/drawings.zip)

brettfo, 2017-10-09:
Can you get a stack trace from the error?  Opening your drawing on both `version-0.2.0` and `version-0.2.1`/`master` succeeded.

Michael-F-Bryan, 2017-10-10:
Sorry, by error I meant parsing returns an `Err(DxfError::UnexpectedEnumValue)`. When I was testing, `unwrap`'s panic message was `the specified enum value does not fall into the expected range`.

I just updated to `0.2.1` from `0.2.0` and tried again on a windows machine (was on Linux when I opened the issue) with a bunch of known-working drawings. Interestingly enough almost all of them failed at the parsing stage with that error.

I'll have a deeper look tonight and see if I can find out where in the DXF parsing process the error is being encountered.

brettfo, 2017-10-10:
I had a little time to look today and the first issue is that the `$DIMZIN` header variable has an unsupported enum value (`8`).  I'm currently working on a proper fix that will allow out-of-range enum values when parsing.  I'll respond here when I have something to show.

brettfo, 2017-10-11:
I've just pushed some updates to `master` that allowed me to fully open your attached drawing.  Can you check with your other drawings?

Michael-F-Bryan, 2017-10-11:
I just tried it out with the latest changes and it looks like it works perfectly with all but one of my drawings :+1:

In the last drawing it returned "invalid digit found in string". This one is a bit different in that was given to us by a customer, so I'm assuming it whatever tool they used to generate it did things slightly differently.

brettfo, 2017-10-11:
Can you share that drawing, possibly privately with me?  If not I understand.

I've pushed more changes that thread file offsets through more error types which should give you a more meaningful error message.  Can you try with the latest from `master` and copy/paste the few lines around the reported error?  You should be able to open the DXF file in a text editor to view the lines.  I might still be able to figure it out from that.

Michael-F-Bryan, 2017-10-12:
In previous crates I've used `error-chain` to chain the errors, allowing you to give the user a high level error message (e.g. "Couldn't parse the drawing"), while also letting people iterate over the causes to find out exactly what went wrong and where. 

This is (roughly) what error messages look like in my library for a different drawing format.

```
Error: Couldn't parse the drawing
  Caused By: Parsing error on line 42, column 7
  Caused By: In drawing body
  Caused By: In Polyline
  Caused By: In Flags
  Caused By: Invalid flag (0x0010)
```

Which is just the output from this helper function:

```rust
fn backtrace(err: &Error) {
  println!("Error: {}", err);
  for cause in err.iter().skip(1) {
    println!("Caused By: {}", cause);
  }
}
```

---

I just updated to the latest `master` and the error message is `invalid digit found in string at line/offset 1`.

[DINO.zip](https://github.com/IxMilia/dxf-rs/files/1378004/DINO.zip)

brettfo, 2017-10-12:
Thanks for the pointer to `error_chain`, I'll start looking into that.

As for `DINO.zip`, I was able to load and `.unwrap()` the contained DXF, both on 1.20 and the recently released 1.21.  Do you have other files that fail to parse?

Michael-F-Bryan, 2017-10-16:
Hehe, turns out my `dino.dxf` test was failing because I'd mixed up file names and the actual contents were in a completely different format.

It looks like `master` works on all the drawings I've got access to :+1:

closed 2017-10-16 by Michael-F-Bryan