ixmilia/dxf-rs #4: error: Specified handle 1 already in use on line 604

bbigras, 2018-04-30:
I tested the dxf with autocad 2002 and 2007.
dxf-rs 35671103fbabc3b5680cda9cbef0f4d39d3eb162

###### error
```
Specified handle 1 already in use on line 604.
Invalid or incomplete DXF input -- drawing discarded.
```
###### code
```rust
extern crate dxf;
extern crate failure;

use failure::Error;

fn test() -> Result<(), Error> {
    use dxf::Drawing;
    use dxf::entities::*;

    let mut drawing = Drawing::default();

    let line = dxf::entities::Line::new(
        dxf::Point::new(0.0, 0.0, 0.0),
        dxf::Point::new(25.0, 40.0, 0.0),
    );
    
    drawing.entities.push(Entity::new(EntityType::Line(line)));
    
    drawing.save_file("file.dxf")?;

    Ok(())
}

fn main() {
    test().unwrap();
}
```

brettfo, 2018-05-04:
Thank you for the bug report!  I don't have a copy of AutoCAD that I can test with but after trying your code I have a theory and I'm hoping you can help me test it.  If you get the opportunity can you do this:

1.  Generate `file.dxf` exactly like you specified above.
2.  Open `file.dxf` in a text editor (e.g., Notepad, etc.) and find the line that contains the value `$HANDSEED`.  There should be only one and my machine generated it at line `402`.
3.  Two lines below `$HANDSEED`, there should be a line that says `1`; **update this value** to `2`.
4.  Save the file and try to open it in AutoCAD.

E.g., change this:

```
$HANDSEED
  5
1
```

to this:

```
$HANDSEED
  5
2
```

If you're curious, the `$HANDSEED` header variable is supposed to be the next available handle, but this is incorrect because the `LINE` entity specifies a handle of `1` on line `604` so the proper value of the next handle is actually `2`.

brettfo, 2018-05-04:
Thank you for testing.  I'll start working on a fix.

brettfo, 2018-05-05:
Apologies for taking so long to get back to the initial issue.  I just pushed a change to master that correctly writes the `$HANDSEED` header variable.

commit dee9fad9 (2018-05-05): write the header after the proper next handle value has been calculated

Fixes #4
closed 2018-05-05 by brettfo