ixmilia/dxf-rs #5: support for 3d polylines

bbigras, 2018-05-04:
It seems dxf-fs doesn't support 3d polylines yet.

brettfo, 2018-05-04:
Can you give an example?  `POLYLINE` should be able to be written like so:

``` rust
use dxf::*;
use dxf::Entities::*;
// ...
let mut poly = Polyline {
    vertices: vec![
        Vertex { location: Point::new(0.0, 0.0, 0.0), .. Default::default() },
        // specify additional vertices
    ],
    .. Default::default()
};
poly.set_is_3d_polyline(true); // this might be the problem
drawing.entities.push(Entity {
    common: Default::default(),
    specific: EntityType::Polyline(poly),
});
```

brettfo, 2018-05-05:
I may have a solution that you can try in the `write-poly` branch in parallel with `poly.set_is_3d_polyline(true)`.  Let me know if that works; if so I'll merge it to `master`.

bbigras, 2018-05-05:
Thanks, I'll try on Monday at work.

bbigras, 2018-05-07:
I still have the same error message.

If it could help here's a template that I used last week to get a working 3d polyline. I have no idea if I did it the right way but it seemed to work.

```jinja
ENTITIES
{%- for line in lines %}
  0
POLYLINE
  5
2B
100
AcDbEntity
  8
{{ line.layer }}
100
AcDb3dPolyline
 66
     1
 10
0.0
 20
0.0
 30
0.0
 70
     8
{%- for point in line.points %}
  0
VERTEX
  5
2C
100
AcDbEntity
  8
0
100
AcDbVertex
100
AcDb3dPolylineVertex
 10
{{ point.x }}
 20
{{ point.y }}
 30
{{ point.z }}
 70
    32{% endfor %}
  0
SEQEND{% endfor %}
  5
2E
100
AcDbEntity
  8
0
  0
ENDSEC
```

brettfo, 2018-05-07:
Thank you for that script, I think I've captured everything in the latest update to the `write-poly` branch.  When you get a chance try it out and let me know if that unblocks you.  If so I'll squash and merge to master.

If you're interested the parts I was missing were different subclass markers for 3d polylines and vertices (`AcDb3dPolyline` and `AcDb3dPolylineVertex`).

bbigras, 2018-05-07:
It works, with:

```rust
vertices: vec![
            Vertex { location: Point::new(0.0, 0.0, 0.0), .. Default::default() },
            Vertex { location: Point::new(25.0, 25.0, 50.0), .. Default::default() },
            Vertex { location: Point::new(50.0, 25.0, 50.0), .. Default::default() },
            Vertex { location: Point::new(50.0, 50.0, 50.0), .. Default::default() },
            Vertex { location: Point::new(25.0, 50.0, 50.0), .. Default::default() },
            Vertex { location: Point::new(25.0, 25.0, 50.0), .. Default::default() },
        ],
```

I get:

[image]

Thanks!

brettfo, 2018-05-07:
Thank you for taking the time to test this.  I've updated `master` with all changes.

commit 36d37e3d (2018-05-07): write the appropriate subclass markers for 3d polylines

Fixes #5
closed 2018-05-07 by brettfo