ixmilia/dxf-rs #7: Saved dxf file cannot be opened by CAD applications

seven-two-eight, 2018-12-15:
After load and save a very simple DXF file with
```Rust
let mut drawing = Drawing::load_file(input)?;
drawing.normalize();
drawing.save_file(output)?;
```
the saved file cannot be opened by QCAD, LibreCAD or ODA File Converter.
The ODA file converter gives error message:

```
ODA Warning: Syntax error or premature end of file on line/offset 1498.
OdError thrown during readFile of drawing "Drawing2.dxf" :
  Object of class AcDbRegAppTable can't be cast to AcDbSymbolTableRecord.
```

brettfo, 2018-12-17:
Thanks for the file!

What version of this code are you using?  Also, what version of the ODA converter?  Latest from `master` with ODA 19.5.0 and 19.8.0 works for me on Windows; I don't have a Linux system to test on.

One thing I did notice that I'll have to fix is that for each vertex in the `LWPOLYLINE` I'm [always outputting the vertex ID, even when it's zero](https://github.com/IxMilia/dxf-rs/blob/23027e397d20ea9b2dbbe54cd6f9a8d1d60451c8/spec/EntitiesSpec.xml#L614) which I probably shouldn't, since the AutoCAD file you attached doesn't have those.  Possibly try opening the DXF saved by this library in a text editor, searching for the string `LWPOLYLINE` and under that you'll see a series of `10` and `20` code pairs which are the X and Y values of the points.  Each one will be accompanied by a code `91` and a `0` on the next line.  Try dropping those.

seven-two-eight, 2018-12-17:
I'm also using ODA 19.8.0 on windows. ODA complains even after I removed the `0` vertex IDs.
It could be an issue related to my version of AutoCAD 2016.
If I create dwg/dxf using LibreCAD, the output of dxf-rs can be directly loaded in QCAD, and also in LibreCAD but with some warning. Only AutoCAD 2016 will throw errors, with or without ODA converter.

Closing the issue because there is at least one functioning workflow.

Thank you for writing dxf-rs!

closed 2018-12-17 by seven-two-eight

brettfo, 2018-12-17:
Glad it's (at least somewhat) useful.  If you find other issues, or find a way this library can write DXF files better, please do let me know, I'd like to make this as usable as possible.