ixmilia/dxf-rs #9: Write error with just readed

happydpc, 2019-10-28:
I have just read the same file and write it out.  But get the error code like these.

$HIDETEXT
280
     1
  9
$HIDETEXT
290
1
  9

$INTERSECTIONDISPLAY
280
     0
  9
$INTERSECTIONDISPLAY
290
0

Compare to the origin file, this occurred only once, but dxf library made it twice. 

And also the Unicode string cant display.

brettfo, 2019-10-28:
For the `$HIDETEXT` and `$INTERSECTIONDISPLAY` issues, have you done a full clean/build?  There's an existing test that verifies these values aren't double-written which I just made more explicit in commit 4c2b873e7cfc185ca859b6340ef2a02733164a24.

brettfo, 2019-10-28:
Regarding the Unicode issues, what version of this library are you using?  DXF files are rather picky about text encoding and I updated this library back in August with proper Unicode support in commit 33e0d68c2bf82d9f04a7822966d3249908a43341.

Do you have any sample files you can share?  If you don't want to post them publicly you can email them to me and I won't expose the contents.  My personal email is `brett.forsgren`@`outlook.com`

happydpc, 2019-10-29:
I find the problem is here:

use encoding::{Encoding, DecoderTrap};
use encoding::all::GBK;
pub(crate) fn read_line<T>(reader: &mut T) -> Option<DxfResult<String>>
where
    T: Read + ?Sized,
{
    let mut bytes = vec![];
    let reader_bytes = reader.bytes();
    for (i, b) in reader_bytes.enumerate() {
        let b = match b {
            Ok(b) => b,
            Err(e) => return Some(Err(DxfError::IoError(e))),
        };
        match (i, b) {
            (0, 0xEF) | (1, 0xBB) | (2, 0xBF) => (), // skip UTF-8 BOM
            _ => {
                if b == b'\n' {
                    break;
                }
                bytes.push(b);
            }
        }
    }

    let newBytes = bytes.clone();
    let mut result = match String::from_utf8(bytes) {
        Ok(s) => s,
        Err(_) =>{
            match GBK.decode(&newBytes, DecoderTrap::Ignore){
                Ok(gbk_s) => gbk_s,
                Err(_) =>{
                    return Some(Err(DxfError::MalformedString));
                }
            }
        }
    };
    if result.ends_with('\r') {
        result.pop();
    }

    Some(Ok(result))
}

happydpc, 2019-10-29:
Also the results of reading these tables are not right. When there exists mutli tables.

```rust
            match &*pair.assert_string()? {
                "APPID" => read_app_ids(drawing, iter)?,
                "BLOCK_RECORD" => read_block_records(drawing, iter)?,
                "DIMSTYLE" => read_dim_styles(drawing, iter)?,
                "LAYER" => read_layers(drawing, iter)?,
                "LTYPE" => read_line_types(drawing, iter)?,
                "STYLE" => read_styles(drawing, iter)?,
                "UCS" => read_ucss(drawing, iter)?,
                "VIEW" => read_views(drawing, iter)?,
                "VPORT" => read_view_ports(drawing, iter)?,
                _ => Drawing::swallow_table(iter)?,
            }
```

brettfo, 2019-10-29:
Can you share a drawing with me so I can better understand what's not working?

happydpc, 2019-10-30:
Ok, I will send to you.

brettfo, 2019-11-12:
Fixed in ec9b113b01f70bc62eb79a7a50d1aa3dcdeaf087 by adding two new methods which in this case allow GBK/GB18030 to be specified as the default text encoding.

``` rust
Drawing::load_with_encoding(<&mut Read>, encoding: Encoding);
Drawing::load_file_with_encoding(filename: &str, encoding: Encoding);
```

closed 2019-11-12 by brettfo