ixmilia/dxf-rs #11: Accept a `Path` when reading/writing files

gdetrez, 2020-09-12:
This is a minor usability issue.  If I have a `Path`, it would be nice be able to do `Drawing::load_file(&mypath)` instead of `Drawing::load_file(&mypath.to_str().unwrap())`.

I think it is possible to do that without breaking the current behavior by using the same trick as `std::fs::File::open` and take the trait `AsRef<Path>` as parameter.  I.e. sth like `Drawing::load_file(file_name: &impl AsRef<Path>)`.  That way, it will still accept a `&str` but also a `Path`, `PathBuf`, etc.

I can give it a go and make a PR if you want.

brettfo, 2020-09-12:
I agree, that sounds really useful. If you're willing to do the work so it'll take `Path` and still `&str` I'd love to take it.

brettfo, 2020-09-30:
Closed via #12.

closed 2020-09-30 by brettfo