ixmilia/dxf-rs #14: UnexpectedCodePair

DGriffin91, 2020-11-03:
When parsing a DXF file using `Drawing::load_file` (Converted by the ODA File Converter) I get this error:
`UnexpectedCodePair([@127890]0/ , "expected 0/ENDTAB")`

The original DWG file (created in Autocad). [Single Lines.zip](https://github.com/ixmilia/dxf-rs/files/5484646/Single.Lines.zip)

I've tried multiple DXF versions, and both 0.4.0 and the github main branch of dxf-rs.

I'm on windows 10, rustc 1.47.0.

Thanks for all your work on this crate!

brettfo, 2020-11-03:
Can you attach the DXF that ODA File Converter generated that you then passed to `load_file`?  I want to make sure I'm looking at the exact same thing.

DGriffin91, 2020-11-03:
Sure can!

[CCLVSingleLines2000.zip](https://github.com/ixmilia/dxf-rs/files/5484688/CCLVSingleLines2000.zip)

brettfo, 2020-11-04:
Thanks for this!

### Short version

When using the ODA converter, save as an ASCII DXF instead of binary and it'll open.

### Long version

I'm reading binary string values incorrectly.  Most strings in binary DXF files are a 1-byte length followed by the ASCII bytes themselves, but object handles (code 340), while a string in ASCII DXF files, _don't_ have a prefixed length and in a binary file are instead a null-terminated C-string, and that throws off the binary reader.  E.g., in this file is the following sequence: `54 01 30 00`.  `54 01` is the little-endian encoding of `340`, the code for an object handle.  At that point I switch into the incorrect string reader mode and read `30` as the length of the following string and then cast all of those to bytes.  Instead, since this is a handle, I need to go into C-string mode which will then return `0\0`, which is the null-terminated string of `"0"` which is an empty object handle ID.

I'll start working on a fix, but it may take a bit to get it properly implemented.

DGriffin91, 2020-11-04:
Thank you for looking into this, I appreciate it.

I am having an issue opening the file in ASCII as well:
[SingleLines2000ASCII.zip](https://github.com/ixmilia/dxf-rs/files/5484896/SingleLines2000ASCII.zip)
I get: `UnexpectedCodePair([@813747]1030/0.0, "expected 0/ENDSEC")` with this file.

brettfo, 2020-11-04:
Thanks, I'll start looking at that, too.

Side note, I need to amend my previous comment; strings in binary DXF files are C-style with a null terminator, but binary chunks are prefixed by a length.  It'll be a bit longer to properly handle that type of data.

brettfo, 2020-11-04:
@DGriffin91 I've just pushed 2 commits to `main` that make it so I can open your original drawing.  Can you give it another shot?

DGriffin91, 2020-11-05:
@brettfo Thank you! This is working now with the 2000 ASCII and Binary version. As well as the 2013 ASCII version. This gets me up and running. Much appreciated.

I do get this error with the 2013 Binary version:
`UnexpectedEnumValue(402428)`

[CCLVSingleLines2013.zip](https://github.com/ixmilia/dxf-rs/files/5491262/CCLVSingleLines2013.zip)

I don't need this to work to be able to move forward, but thought it may be helpful.

brettfo, 2020-11-05:
Thanks for being so thorough in your testing.  I've pushed one more commit that allows your most recent file to be read successfully.

DGriffin91, 2020-11-05:
You're welcome! Glad to help. Just tested this latest commit and it is working. Thanks again!

closed 2020-11-05 by DGriffin91