ixmilia/dxf-rs #16: Some DXF file make dxf2json unhappy and panic

piaoger, 2020-12-29:
Some DXF files make dxf2json unhappy and panic. Please see dxf file in attachments:

- panic for differenct reasons:

```
 
`Err` value: UnexpectedCodePair([@252559]1030/0.0, "expected 0/ENDSEC")'
`Err` value: UnexpectedCode(....)'
`Err` value: ImageError(IoError(Custom { kind: UnexpectedEof, error: "failed to fill whole buffer" }))

```
[dxfs.zip](https://github.com/ixmilia/dxf-rs/files/5750181/dxfs.zip)

- Below dxf Can be converted after running "recover" in ODADrawingExplorer

01.dxf ,  02.dxf , 04.dxf, 07.dxf

- I attached both origin dxf and fixed ones in the zip file.

brettfo, 2021-01-04:
The first error you posted was fixed in commit 696d4ce8df84cbab50bc8c99e03fbbfb7a387d07 which is now in `main` (but not the crate), but files 04, 06, 07, and 08 have issues with the thumbnail data, so I'll start to look at those.

brettfo, 2021-01-04:
I just pushed commit ea2f233f69c6444cdf9d6ed16d96d7cd3878aa33 which fixes the remaining files.  I was incorrectly calculating the thumbnail palette size which meant the calculated file header was wrong.

closed 2021-01-04 by brettfo