ixmilia/dxf-rs #18: A DXF file containing 3D cannot be opened after saving as a new dxf

piaoger, 2021-01-23:
I tried a couple of DXF files containing 3D and most of them are work fine though not perfect. For below one, the rewritten file cannot opened in ODA.

## Origin in ODA:

[image]

## Save as a new dxf

```
let drawing = Drawing::load_file("HH016.dxf").unwrap();
drawing.save_file("HH016.rewrite.dxf").unwrap();
```

The file size is much smaller and a recovery messagebox is prompted in ODA.

- Need to recover while loading in ODA again

[image]

- Error message in ODA console
 
```
Enter name of drawing to open: "/Users/goodman/Downloads/dxf-files/HH016.rewrite.dxf"
ODA Warning: AcDbRegAppTableRecord(3A29E628): Duplicate record name "ACAD_MLEADERVER" - Ignored

ODA Warning: Syntax error or premature end of file on line/offset 1128.

Error: Object of class AcDbRegAppTableRecord can't be cast to AcDbBlockTableRecord.
Error: Loading of file "/Users/goodman/Downloads/dxf-files/HH016.rewrite.dxf" was failed.
Command: recover "/Users/goodman/Downloads/dxf-files/HH016.rewrite.dxf"
Error Recovering Database...Object of class AcDbRegAppTableRecord can't be cast to AcDbBlockTableRecord.
```

## Origin and rewritten dxf files

[dxf-files.zip](https://github.com/ixmilia/dxf-rs/files/5860019/dxf-files.zip)

brettfo, 2021-01-26:
A quick check shows that the input file uses a `u64` for handle values and I currently only support `u32` so the values are getting truncated so there are duplicate values. Checking the spec shows that `u64` is correct.

I'll start working on fixing my handle handling.

brettfo, 2021-01-26:
Commit c58f936ba9f6d6eb4e45eb4f6aba1beb31985cdd in `main` treats handles as `u64`, now.  I was able to use your sample and open the `.rewrite.dxf` file in ODA.  Opening that drawing in AutoCAD caused it to crash because AutoCAD is very restrictive in what it will accept; compatibility with AutoCAD is an ongoing struggle.

closed 2021-01-26 by brettfo

piaoger, 2021-01-26:
The result file just crashes ODA using your latest change though it can be opened in QCAD. 

- env

```
ODAFileConverter_QT5_macOsX_x64_10.14dll_21.11.dmg
Mac OSX
``` 
- my test code

```
extern crate dxf;
extern crate serde;
extern crate serde_json;

use dxf::Drawing;
use std::env;
use std::fs::File;
use std::io::{BufWriter, Write};

use std::time::Instant;

fn main() {

    let args: Vec<String> = env::args().collect();

    println!("- loading drawing");
    let dxf_path = &args[1];
    let now = Instant::now();
    let drawing = Drawing::load_file(&dxf_path).unwrap();
    println!("  cost time(loading): {:?}", now.elapsed());

    println!("- saving drawing to new file");
    let now = Instant::now();
    let mut rewrite_drawing_path = dxf_path.clone();
    rewrite_drawing_path.push_str(".rewrite.dxf");
    drawing.save_file(&rewrite_drawing_path);
    println!("  cost time(rewrite): {:?}", now.elapsed());

    println!("- saving drawing to json");
    let now = Instant::now();
    let mut json_path = dxf_path.clone();
    json_path.push_str(".json");
    let json_file = File::create(&json_path).unwrap();
    let mut writer = BufWriter::new(json_file);
    let drawing_json = serde_json::to_string_pretty(&drawing).unwrap();
    writer.write_all(drawing_json.as_bytes()).unwrap();
    println!("  cost time(to_json): {:?}", now.elapsed());

    println!("- loading new drawing from json");
    let now = Instant::now();
    let new_drawing :Drawing = serde_json::from_slice(drawing_json.as_bytes()).unwrap();
    println!("  cost time(from_json): {:?}", now.elapsed());

    println!("- saving new drawing");
    let mut new_drawing_path = dxf_path.clone();
    new_drawing_path.push_str(".json.dxf");
    new_drawing.save_file(&new_drawing_path);
    println!("  cost time(rewrite_json): {:?}", now.elapsed());
}

```

- Neither HH016.dxf.json.dxf nor HH016.dxf.rewrite.dxf can be opened in ODA properly.
  But it can be opened in ODA after conversion with ODA converter.