ixmilia/dxf-rs #19: Trouble with Russian letters

ThisIsRex, 2021-05-23:
I have a trouble with some Russian letters, they are displayed as "?"
[image]

in dxf it looks like this (^X instead of Unicode codes):
`^P^Q^R^S^T^U^A^V^W^X^Y^Z^[^\^]^^^_\U+0420^^\U+0421\U+0422\U+0423\U+0424\U+0425\U+0426\U+0427\U+0428\U+0429\U+042A\U+042B\U+042C\U+042D\U+042E\U+042F`
`\U+0430\U+0431\U+0432\U+0433\U+0434\U+0435\U+0451\U+0436\U+0437\U+0438\U+0439\U+043A\U+043B\U+043C\U+043D\U+043E\U+043F\U+0440\U+043E\U+0441\U+0442\U+0443\U+0444\U+0445\U+0446\U+0447\U+0448\U+0449\U+044A\U+044B\U+044C\U+044D\U+044E\U+044F`

brettfo, 2021-05-24:
Do you have a DXF file you can share with these values?  If you don't want to post a sample publicly you can email it directly to me at `brett.forsgren` `@` `outlook.com` and I'll keep the contents private.  And can you give me an example of what you expect the string value to be?  In the sample you gave all of the `\U+xxxx` correspond to a Unicode character and appear to be handled correctly, so it's just the `^` characters that are displaying as `?`, but according to [the spec](https://help.autodesk.com/view/OARX/2020/ENU/?guid=GUID-77D34F88-73CC-495D-B48C-37628EE61D73), all values of the form `^#` are simply ASCII control characters, which don't have a display value, hence the '?'.  In this example, the `^P` through `^_` correspond to the following HEX values:

```
^P => 0x10
^Q => 0x11
^R => 0x12
^S => 0x13
^T => 0x14
^U => 0x15
^A => 0x01
^V => 0x16
^W => 0x17
^X => 0x18
^Y => 0x19
^Z => 0x1A
^[ => 0x1B
^\ => 0x1C
^] => 0x1D
^^ => 0x1E
^_ => 0x1F
```

That sequence is a simple progression from the value `0x10` through `0x1F` with the exception of the random `0x01` (`^A`) thrown in the middle, so I'm wondering if that progression has some meaning that I'm not decoding appropriately.

brettfo, 2021-05-24:
A quick thing I just noticed is that the letters appear to be in alphabetical order which means the letter `A` is probably expected to come first and it's Unicode value _should_ be `\U+0410`, but instead the file contains `^P` = `0x10` which makes me think the leading byte of `0x04` got dropped somewhere.  If that's the case can you tell me how this file was created?  If it was done with this library can you share a code sample that generates this issue?

brettfo, 2021-05-24:
I think I figured it out and fixed it in commit 8fcb046e704ae2ee596dab988dd14451e83b6a7e.  The location of the "real" fix was [this](https://github.com/ixmilia/dxf-rs/commit/8fcb046e704ae2ee596dab988dd14451e83b6a7e#diff-996287da739a309ed6acc600bc0cbbc1122e89089157c265025ed56e42b88000R130) where I was incorrectly checking for the ASCII control escaping by narrowing to 8 bits instead of 32.

ThisIsRex, 2021-05-24:
The code:
```
#![feature(default_free_fn)]
use std::default::default;

use dxf::Drawing;
use dxf::{entities::*, Point};

const A1_WIDTH: f64 = 841.0;
const A1_HEIGHT: f64 = 594.0;

fn main() {
    let mut drawing = Drawing::default();

    drawing.entities.push(Entity::new(EntityType::Text(Text {
        value: "АБВГДЕЁЖЗИЙКЛМНОПРСТУФХЦЧШЩЪЫЬЭЮЯ".to_string(),
        text_height: 40.0,
        location: Point::new(0.0, A1_HEIGHT - 45.0, 0.0),
        ..default()
    })));
    drawing.entities.push(Entity::new(EntityType::Text(Text {
        value: "абвгдеёжзийклмнопрстуфхцчшщъыьэюя".to_string(),
        text_height: 40.0,
        location: Point::new(0.0, A1_HEIGHT - 105.0, 0.0),
        ..default()
    })));

    drawing.save_file("test.dxf").expect("save error");
}
```
Expected (uppercase):
`\u0410\u0411\u0412\u0413\u0414\u0415\u0401\u0416\u0417\u0418\u0419\u041A\u041B\u041C\u041D\u041E\u041F\u0420\u0421\u0422\u0423\u0424\u0425\u0426\u0427\u0428\u0429\u042A\u042B\u042C\u042D\u042E\u042F`
and (lowercase):
`\u0430\u0431\u0432\u0433\u0434\u0435\u0451\u0436\u0437\u0438\u0439\u043A\u043B\u043C\u043D\u043E\u043F\u0440\u0441\u0442\u0443\u0444\u0445\u0446\u0447\u0448\u0449\u044A\u044B\u044C\u044D\u044E\u044F`

I've attached the file
[test.zip](https://github.com/ixmilia/dxf-rs/files/6534015/test.zip)

ThisIsRex, 2021-05-24:
> 
> 
> I think I figured it out and fixed it in commit [8fcb046](https://github.com/ixmilia/dxf-rs/commit/8fcb046e704ae2ee596dab988dd14451e83b6a7e). The location of the "real" fix was [this](https://github.com/ixmilia/dxf-rs/commit/8fcb046e704ae2ee596dab988dd14451e83b6a7e#diff-996287da739a309ed6acc600bc0cbbc1122e89089157c265025ed56e42b88000R130) where I was incorrectly checking for the ASCII control escaping by narrowing to 8 bits instead of 32.

this works, thanks!

closed 2021-05-24 by brettfo