ixmilia/dxf-rs #21: Fail parsing on empty $FINGERPRINTGUID and $VERSIONGUID

kurtlawrence, 2021-07-10:
Hi there,

I am receiving a parsing error for the attached DXF file. 
[out1.zip](https://github.com/ixmilia/dxf-rs/files/6794737/out1.zip)

If I remove the lines 950-957 it successfully parses. The contents of those lines is as follows:
```
$FINGERPRINTGUID
  2

  9
$VERSIONGUID
  2

  9
```

I am wondering if it is possible to have a loading mechanism that is tolerant of these empty entries?

kurtlawrence, 2021-07-10:
Looks like it is related to: https://github.com/IxMilia/Dxf/issues/83

brettfo, 2021-07-12:
Good call, and you're right, the fix was exactly the same as the C# version of this library.  I've pushed a commit that covers this case.

commit 76c334dc (2021-07-12): default to a nil UUID if the supplied value can't be parsed

Fixes #21.
closed 2021-07-12 by brettfo