ixmilia/dxf-rs #23: Can't create block successfully

Dynabits, 2021-11-19:
Every Time ACAD 2018 crashes with the generated file from this code:

```
    let mut drawing = Drawing::new();

    let point = ModelPoint::default();
    let point = Entity::new(EntityType::ModelPoint(point));
    // drawing.add_entity(point);

    let mut att = AttributeDefinition::default();
    att.text_tag = String::from("tag");
    att.prompt = String::from("tag_prompt");
    att.value = String::from("test_value");
    att.value = String::from("1");
    let att = Entity::new(EntityType::AttributeDefinition(att));
    // drawing.add_entity(att);

    let mut lyr1 = Layer::default();
    lyr1.name = String::from("layer1_test");
    drawing.add_layer(lyr1);

    let mut blc = Block::default();
    blc.name = String::from("blc_name");
    blc.set_has_non_consistent_attribute_definitions(true);
    let mut vec = Vec::new();
    vec.push(att);
    vec.push(point);
    blc.entities = vec;
    drawing.add_block(blc);

    drawing.save_file("test.dxf").ok();
```

I think I'm not creating the block correctly, the goal is to have the point and the attribute entities inside the block.

Any help would be very appreciated.

PS: I'm new to rust and dxf spec.

brettfo, 2021-11-19:
AutoCAD is _extremely_ pick about what it will accept, and frankly I haven't figured out the exact formula yet.  That said, other CAD apps are less strict and LibreCAD will open the following DXF with a block:

``` rust
let mut drawing = Drawing::new();

// this creates a block with a single line from (0,0) to (1,1)
// n.b., the block will not yet be displayed on screen; it exists simply as a "template" that can be inserted later
let mut block = Block::default();
block.name = String::from("block_name");
block.entities.push(Entity::new(EntityType::Line(Line::new(
    Point::new(0.0, 0.0, 0.0),
    Point::new(1.0, 1.0, 0.0),
))));
drawing.add_block(block);

// this inserts a copy of the "block_name" block at location (2,2)
// the result of this is a line from (2,2) to (3,3)
let mut ins = Insert::default();
ins.name = String::from("block_name");
ins.location = Point::new(2.0, 2.0, 0.0);
let ins = Entity::new(EntityType::Insert(ins));
drawing.add_entity(ins);

// this inserts a copy of the "block_name" block at location (2,-2)
// the result of this is a line from (2,-2) to (3,-1)
let mut ins = Insert::default();
ins.name = String::from("block_name");
ins.location = Point::new(2.0, -2.0, 0.0);
let ins = Entity::new(EntityType::Insert(ins));
drawing.add_entity(ins);

drawing.save_file("test.dxf").ok();
```

Dynabits, 2021-11-20:
@brettfo thank you for your reply and example code, when I run it, the same result ACAD say it's invalid drawing.

Maybe someone like you with the required expertise can look at other libraries like [link](https://ezdxf.readthedocs.io/en/stable/dxfinternals/block_management.html) and see what could be the problem?

brettfo, 2021-11-29:
@Dynabits Thank you for that link, I hadn't seen that before and it was very helpful.

I've pushed commit 944f6e4d1932da73e886a11c75e33bfa25b54bb3 to `main` which will write a `BLOCK`/`INSERT` in R12 and I can then open it in AutoCAD 2016 (that's the only version I have to test with.)

That commit also contains the file `examples/src/block_examples.rs` which should get you started.

I'll start testing R13+ and see what I can come up with.