ixmilia/dxf-rs #25: Support Hatch entities

kurtlawrence, 2022-05-09:
Hi there,

Would it be possible to get support for hatch entities?
This output comes from QGIS.

[example.zip](https://github.com/ixmilia/dxf-rs/files/8655865/example.zip)

brettfo, 2022-05-10:
`HATCH` entities are really complex and while eventually I'll add support for them, it's not something that'll likely happen soon, purely due to lack of time.

I'll keep this issue open, though, until they are implemented.

guradia, 2022-10-19:
Aye!

I just came around the need to work with DXF-Files including Hatches. At least two different usages are there to play with and currently i'm just seeking to parse them.
Some pointers would be nice as to where to start implementing them inside the library.

XML-Definitions for a generator and Structs that could be generated or handcrafted are easy to stumble upon ... but the intricate coupling of things.

Perhaps an overview of how a simple Line-Entity journeys from spec to usable form, could be written down without too much hassle for you.

brettfo, 2022-12-03:
I also maintain a C#/.NET DXF reader/writer that has limited support for `HATCH` entities.  The relevant sources are:

- [XML definition](https://github.com/ixmilia/dxf/blob/7188fa418dee5f338790e2258cff992496afeddf/src/IxMilia.Dxf.Generator/Specs/EntitiesSpec.xml#L534)
- [source definition 1](https://github.com/ixmilia/dxf/blob/7188fa418dee5f338790e2258cff992496afeddf/src/IxMilia.Dxf/Entities/DxfHatch.cs)
- [source definition 2](https://github.com/ixmilia/dxf/blob/7188fa418dee5f338790e2258cff992496afeddf/src/IxMilia.Dxf/Entities/DxfHatch.PatternDefinitionLine.cs)
- [source definition 3](https://github.com/ixmilia/dxf/blob/7188fa418dee5f338790e2258cff992496afeddf/src/IxMilia.Dxf/Entities/DxfHatch.BoundaryPath.cs)

jannewulf, 2024-06-28:
Are there any plans to add support for this?

brettfo, 2024-06-29:
@jannewulf I mentioned in [this comment](https://github.com/ixmilia/dxf-rs/issues/25#issuecomment-1122736023) that eventually I'd like to add support for them, but there is no timeline on that.

DennisJensen95, 2024-12-03:
This would be awesome, i tried 30 minutes of trying to wrap my head around it and implemented some of the XML spec and generated the code. But I gave up and opted out of Hatch for now, would be lovely to implement it, but fully understand that time is constrained resource 👍

DennisJensen95, 2025-08-01:
I do not any alternatives. I would however at some point implement the hatch 😊

samuelint, 2025-08-01:
I need to export a polygon that have holes in it. @DennisJensen95 what is your progress on the implementation?

DennisJensen95, 2025-08-01:
Progress is Zero 😂😂

I have implemented some other stuff though. So I know how to work around the lib 😊

samuelint, 2025-08-01:
What is the effort to implement this? What are the challenges making it hard to implement?

DennisJensen95, 2025-08-01:
I have not looked at it yet. So I would not know. I think the only way to properly know is to take a stab at it. Add it to the Entities.xml file and generate it. It might be that the hatch needs it's own parser and the spec does not suffice, not sure. Do you want to take a stab at it, I can guide a bit if you need? 😊

samuelint, 2025-08-02:
Here a pull-request:
https://github.com/ixmilia/dxf-rs/pull/102
I've used Claude Code and it has nailed it!

dariush624, 2026-05-30:
Is this still up to date? It'd be nice to have it