ixmilia/dxf-rs #26: Unable to save as R2013

WCarrollSTO, 2022-05-18:
When I open and save (with no modifications) a blank file, I am unable to open it in AutoCAD.

Details:
- Using AutoCAD LT 2022 
- Version R2013
- Blank File (just using the default AutoCAD template)
- No modifications made in code
- I have tried running `normalize` and not; neither yield a difference

Code used:
```rust
let table = a.asm_table();
let dxf = Drawing::load_file(&in_path).unwrap();

let mut buf = Vec::new();
dxf.save(&mut buf).unwrap();
write(path, buf)
```

Error: 
```
Bad handle 0: already in use
Error in APPID Table
DXF read error on line 1718.
Invalid or incomplete DXF input -- drawing discarded. 
```
Blank File generated from AutoCAD: [Blank.dxf.txt](https://github.com/ixmilia/dxf-rs/files/8717496/Blank.dxf.txt)
This same file, but open and save (without modification): [LoadSave.dxf.txt](https://github.com/ixmilia/dxf-rs/files/8717654/LoadSave.dxf.txt)

Here is an entirely blank file made with this library, only difference is changing version after `Drawing::new()` to `AcadVersion::R2013`:
[UsingDrawing::New.dxf.txt](https://github.com/ixmilia/dxf-rs/files/8717493/RF1-1Out.dxf.txt)
 

Please let me know if there is any further info you need.  Thank you!

tr3ysmith, 2024-01-18:
I have this same problem, trying to do any other version other than R12

R14 or R2018 doesn't work, both showing this same error

anscarlett, 2024-06-21:
Is this project dead? I don't see any updates for quite some time.

I tried conversion using the ODA file converter which will convert a R2018 file just fine, the original output file is ~16k, with the ODA file coming in at ~74k

brettfo, 2024-06-21:
> Is this project dead?

Not dead, but something that I spend my free time on, which can vary quite a bit.

The R12 DXF format is reasonably simple, but from R13/R14 on it gets really complicated and I simply haven't had the chance to work my way through all of the issues.  By far the most complicated issue is object handles.  The spec from Autodesk doesn't properly explain all of it so the process to fix these issues is simple experimentation.

If you're really in need of converting files, the ODA converter is certainly a good option and that's one of my primary ways of testing file compatibility, but even then, there are differences between AutoCAD and ODA in what each will accept.

anscarlett, 2024-06-24:
> > Is this project dead?
> 
> Not dead, but something that I spend my free time on, which can vary quite a bit.

Appreciate the response, and I totally understand the issues. I'm quite interested in utilising some features in later versions (R2004), specifically using rust.

I've been digging through other documentation to get a grip on this. Any pointers you have would be appreciated.

brettfo, 2024-06-29:
A large part of the code for this is auto-generated at build time.  Look at the `build/` directory for the code that does this as well as the `spec/` directory for the XML files that `build/` uses.

A lot of the work is getting a file from AutoCAD and ensuring this library can read it and once that's done, the reverse; create a small app that writes a file and see if AutoCAD can open it appropriately.  AutoCAD is very specific about what it will accept and their error messages are near useless.  I've had some better luck first making ODA happy and then moving on to AutoCAD.