ixmilia/dxf-rs #27: UnexpectedCode Err on $ACADMAINTVER

usagi, 2022-07-28:
## Err 

- `thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: UnexpectedCode(90, 11)', src/main.rs:5:43`

```rs
let o = dxf::Drawing::load_file("a.dxf").unwrap(); // <- src/main.rs:5
```

## Problem

This a.dxf contains `$ACADMAINTVER`. A part of it is shown below:

```dxf
︙
HEADER
  9
$ACADVER
  1
AC1032
  9
$ACADMAINTVER
 90
       55
  9
$DWGCODEPAGE
  3
︙
```

Note that the library worked as expected when 4 lines were removed from `$ACADMAINTVER`.

usagi, 2022-07-28:
Additional info:

- "a.dxf" was output by `ODAFileConveter` in `ACAD2018` configuration. The Group Code in this `$ACADMAINTVER` section is 90.
- In `ACAD2013` or earlier version settings the code in the `$ACADMAINTVER` section is 70.

And I was run with debugger, then I found the code in "target/debug/build/dxf-9a553b685b0574c6/out/generated/headers.rs" L1471:

```rs
            "$ACADMAINTVER" => { verify_code(&pair, 70)?; self.maintenance_version = pair.assert_i16()?; },
```

And, the specification indeed states that it is 70 => https://help.autodesk.com/view/OARX/2018/ENU/?guid=GUID-A85E8E67-27CD-4C59-BE61-4DC9FADBE74A . I do not know if this library should allow 90 or if ODAFileConverter should output 70 even in DXF2018. 🤔

brettfo, 2022-10-28:
While the docs report that `$ACADMAINTVER` should be ignored, I'd like to keep the field in case somebody needs to use it.

I've pushed commit 17265bf519dbfed4d281a12996735ac6380c0ad5 to `main` that will allow reading `$ACADMAINTVER` as a code 90, but internally it's still stored as `i16` and only written as code 70.

closed 2022-10-28 by brettfo