ixmilia/dxf-rs #35: problematic read

ykong1337, 2023-02-23:
There is a dwg file, I use ```ODAFileConverter``` to convert it to a dxf file corresponding to the ACAD version.
CadVersion: AC1014

Code used:
```
use dxf::Drawing;

fn main() {
    run();
}

pub fn run() {
    if let Some(p) = rfd::FileDialog::new()
        .add_filter("dxf", &["dxf"])
        .pick_file()
    {
        let _drawing = Drawing::load_file(p.as_path()).unwrap();
    }
}

```
Error:
```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: UnexpectedCodePair([@322045]340/2F, "expected 0/ENDSEC")'
```
[fj.dwg.txt](https://github.com/ixmilia/dxf-rs/files/10809619/fj.dwg.txt)
[fj.dxf.txt](https://github.com/ixmilia/dxf-rs/files/10809620/fj.dxf.txt)

ykong1337, 2023-02-28:
I found that because the file has
```
102
VTR_0.000_0.000_1.000_1.000_VISUALSTYLE
340
2F
102
VTR_0.000_0.000_1.000_1.000_GRIDDISPLAY
  70
      3
102
VTR_0.000_0.000_1.000_1.000_GRIDMAJOR
  70
      5
102
VTR_0.000_0.000_1.000_1.000_DEFAULTLIGHTING
280
      1
102
VTR_0.000_0.000_1.000_1.000_DEFAULTLIGHTINGTYPE
  70
      1
102
VTR_0.000_0.000_1.000_1.000_BRIGHTNESS
141
0.0
102
VTR_0.000_0.000_1.000_1.000_CONTRAST
142
0.0
   0
```
Did you not deal with it?

brettfo, 2023-03-05:
Fixed in 51f49bec082038e3e45c8d7cad7ab4a9b28a2454.

closed 2023-03-05 by brettfo