ixmilia/dxf-rs #45: Is a closed polyline missing nodes

c5cle, 2023-06-21:
This is a closed polyline rectangle, but there are only 4 nodes in the read data, missing the last closed node
[image]

c5cle, 2023-06-21:
This uses files from dxf 2013
[test2013.zip](https://github.com/ixmilia/dxf-rs/files/11811728/test2013.zip)

ykong1337, 2023-06-29:
> This is a closed polyline rectangle, but there are only 4 nodes in the read data, missing the last closed node [image]

Note members: ```flags```
When it is 1 it means closure, you need to draw closure yourself. ``` 1 -> 2 -> 3 -> 4 -> 1 ```
You can ignore it when it is 0. ``` 1 -> 2 -> 3 -> 4 ```

c5cle, 2023-08-19:
> > This is a closed polyline rectangle, but there are only 4 nodes in the read data, missing the last closed node [image]
> 
> Note members: `flags` When it is 1 it means closure, you need to draw closure yourself. `1 -> 2 -> 3 -> 4 -> 1` You can ignore it when it is 0. `1 -> 2 -> 3 -> 4`

thanks.

closed 2023-08-19 by c5cle