ixmilia/dxf-rs #58: Polyface Mesh not exporting correctly

tr3ysmith, 2024-01-18:
Hey, I'm trying to export a Polyline with polyface mesh vertices, and inside rust I can println!() the polyline and it looks right, but the exported dxf isn't correct. I re-imported the DXF and serialized it to json and saw that it was just listed as a normal vertex:

Here's the rust println!() of my polyface mesh
```
(
            Vertex {
                location: Point {
                    x: 0.0,
                    y: 0.0,
                    z: 0.0,
                },
                starting_width: 0.0,
                ending_width: 0.0,
                bulge: 0.0,
                flags: 128,
                curve_fit_tangent_direction: 0.0,
                polyface_mesh_vertex_index1: 2,
                polyface_mesh_vertex_index2: 3,
                polyface_mesh_vertex_index3: -4,
                polyface_mesh_vertex_index4: 0,
                identifier: 0,
            },
            Handle(
                22,
            ),
),
```

And here's the json of the parsed dxf that was exported:
```json
{
                    "location": {
                      "x": 0.0,
                      "y": 0.0,
                      "z": 0.0
                    },
                    "starting_width": 0.0,
                    "ending_width": 0.0,
                    "bulge": 0.0,
                    "flags": 160,
                    "curve_fit_tangent_direction": 0.0,
                    "polyface_mesh_vertex_index1": 0,
                    "polyface_mesh_vertex_index2": 0,
                    "polyface_mesh_vertex_index3": 0,
                    "polyface_mesh_vertex_index4": 0,
                    "identifier": 0
}
```

tr3ysmith, 2024-01-18:
It also looks like `set_is_3d_polygon_mesh` has no effect either, I can see it in the struct before I export but its not actually applied to the output DXF (should be a flag value of 64)

brettfo, 2024-01-20:
Fixed in 4e940ec7f1ac1dea37da1b58a1279ad764bc472e

closed 2024-01-20 by brettfo