ixmilia/dxf-rs #64: Question: why build most of api from XML in build.rs?

mrghosti3, 2024-03-14:
As per the title, I want to know why this library is designed as such to build most of api from XML spec.

brettfo, 2024-03-14:
There's a lot of code that would be duplicated and it's not uncommon to need to update how a value is read/written, so rather than updating hundreds of lines, it's easier to update an XML file and a little bit of the codegen.

As a quick example, `HeaderVariablesSpec.xml` is 325 lines long, and the generated file is 3335, so there's a whole order of magnitude difference.  `EntitiesSpec.xml` is 1341 lines, generated is 4737.

brettfo, 2024-03-14:
A good example is recent PR #56.  It changes a lot of how a test-only JSON parsable version is generated.  Rather than manually change thousands of lines, that PR is only 150 lines long.

mrghosti3, 2024-04-05:
Understood. That explains it then. Gonna play around with macros in maybe replacing build script.

closed 2024-04-05 by mrghosti3