ixmilia/dxf-rs #67: Release situation

MrMuetze, 2024-07-01:
Hey everyone!

I am interested in using this crate for a bigger project and I was wondering what the release situation is currently. Since there is still activity and PRs are being merged, I wonder if there is a chance to get new releases in the future or if there is an underlying issue (maybe with repo user rights or something along those lines) since the last release is quite old now.

Maybe we can add some info to the readme regarding this as well to keep interested people well informed about the situation of this crate! :)

brettfo, 2024-11-09:
This is a fun side project of mine and unfortunately not professionally funded, so the merge and release schedule is really whenever I think about it.

Given how long it's been since a release and that there have been non-trivial changes, I'll bump the version and queue another one.

brettfo, 2024-11-09:
Version 0.6.0 is live on crates.io.

closed 2024-11-09 by brettfo