ixmilia/dxf-rs #85: Can't open DXF file in AutoCAD LT 2025

DennisJensen95, 2024-12-09:
Hi, 

I am trying to open my DXF file i have made with this SDK. But I keep getting this error: 

```bash
Bad handle 0: already in use
Error in APPID Table
DXF read error on line 1120.
Invalid or incomplete DXF input -- drawing discarded.
```

I think it is because by default not all APPIDs for example has a handle. I am trying to convert it to R2018. Any help would be appreciated to fix it. I have tried to look at the github.com/ixmilia/dxf Dotnet package, but no luck yet. 

I would like to fix it, but i think i need a pointer from you @brettfo 😊

DennisJensen95, 2024-12-09:
Oh, i found your todo that i think i need to handle here: https://github.com/ixmilia/dxf-rs/blob/main/build/table_generator.rs#L370. I am going to try and implement it, thinking i can get new handles from the drawing and add them here?

DennisJensen95, 2024-12-10:
I am encountering quite a few things trying to make it compatible with R2018, @brettfo  do you have any recommendations for executing this work? My current feedback loop is, get an error from AutoCAD when loading the R2018, then check the error look at the spec here: https://help.autodesk.com/view/OARX/2024/ENU/?guid=GUID-235B22E0-A567-4CF6-92D3-38A2306D73F3 and then trying to implement it and hit next thing.

DennisJensen95, 2024-12-10:
So far the work has included the following:
- Put handles on tables 
- Write up empty tables
- Adjusting maintenance version to accomodate for code 90.

Now i am working with this error: 

```text
The following error was encountered while reading
in SEQEND starting at line 2306:
Class separator for class AcDbEntity expected
Invalid or incomplete DXF input -- drawing discarded.
```

😊

DennisJensen95, 2024-12-10:
I kinda got stuck at dictionary elements, AutoCAD saying that it needs NamedObjects dictionary... 🤔

brettfo, 2024-12-10:
Compatibility with AutoCAD is a large ongoing issue, but I have a C# DXF handler that has a little bit better compatibility that might offer some hints.  That can be found here: https://github.com/ixmilia/dxf

DennisJensen95, 2024-12-11:
Thank you, I was looking at that repo and also ezdxf python lib. I managed to get compatability with AutoCAD, which was awesome 🚀

At least with AutoCAD LT 2025. Anyone knows how certain one can be that it works with other versions as well?

Dexus, 2025-01-06:
Do you have the chance to commit your changes to your fork? Then I would like to checkout and try get involved.

DennisJensen95, 2025-01-07:
Sure, it is already committed. I did a bunch of stuff so there is more than just this ticket. I have not had time to make PR 😅

Fork: https://github.com/Digitalthaandvaerk/dxf-rs

Dexus, 2025-01-14:
I have tested it with AutoCad 2021 works.

DennisJensen95, 2025-01-15:
That is great!

jallejulius, 2025-04-11:
I still have no luck with opening dxf files created by this crate in major software like Autocad nor Microstation. 
Librecad, DWG fastview and conversion tools works great tho.
@DennisJensen95 what version, or what can I do to increase my odds here? Btw I have tried your fork as well but the problems remains. I cant open any files, even empty ones in DWG trueview either?

DennisJensen95, 2025-04-11:
What AutoCAD version are you running and what types are you generating? 😊

jallejulius, 2025-04-12:
@DennisJensen95 Thank you for your contribution and fast reply!
I was using DWG Trueview 2026. However I finally found what was causing the issue, and that was when I had a space in the layer.name field. When so I couldnt open the file in DWG Trueview.

```
    pub fn test_function(&mut self) {
        let mut dxf_file = Drawing::new();
        
        // let layer_name = "Text 0".to_string(); // This doesnt let me open the file in DWG Trueview
        let layer_name = "Text".to_string(); // This work
        let mut layer = Layer::default();
        layer.name = layer_name.clone();

        let mut block = Block::default();
        block.name = layer_name.clone();
        block.layer = layer.name.clone();

        dxf_file.add_block(block);

        let mut insert = Insert::default();
        insert.name = layer_name.clone();

        let insert_entity = Entity::new(EntityType::Insert(insert));
        dxf_file.add_entity(insert_entity);

        dxf_file.save_file("dxf_file.dxf")
            .expect("Failed to save DXF file");
    }
```

jallejulius, 2025-04-12:
Another thing that I noticed was that I couldnt open a dxf file in DWG Trueview if it contained f64::NAN values.

```
            let mut polyline = Polyline::default();

            // Add vertices using `add_vertex`
            for line in &line_object.line {
                // This check must be done to prevent autocad to ignore/not open the file
                if line[0].is_nan() | line[1].is_nan() {
                    continue;
                }

                let vertex = Vertex {
                    location: Point::new(line[0] - a_distance, line[1], 0.0), // Assume Z = 0.0
                    ..Default::default()
                };
                polyline.add_vertex(dxf_file, vertex);
            }
```